amperka/ino #221: Can a "USBTiny" device (ie. Adafruit Pro Trinket) be uploaded to?

jps3, 2015-01-02:
I have been learning and using inotool for about a week now, to move away from the Arduino.app GUI, as I'm more comfortable with cli tools. I've successfully been able to upload to an Adafruit Pro Trinket only via an FTDI cable (ie. creates a serial port), but not via USB. I have made sure I have performed the step to put the device into a ready state (bootloader?) by pushing the reset button, etc. (Note: I had already updated `boards.txt` file per Adafruit’s instructions for Pro Trinket.)

Last night I was able to upload `firmware.hex` using `avrdude`, but I cannot figure out how to get `ino upload` to do this, as it always give an error not being able to find a `/dev/tty.usbmodem*` or `/dev/tty.usbserial*` device. The `avrdude` command which worked was: `avrdude -c usbtiny -p m328p -D -U flash:w:.build/protrinket3/firmware.hex:i`

Am I just missing something in the `ino` command itself, or doesn't it support uploading to a usbtiny device?

simon-whitehead, 2015-02-04:
Heh - I am also wondering this. I have tried all night with no luck. I received this board as a gift and I __really__ despise the Arduino IDE and want to keep using `vim` - so I hope there is an answer for this :/

jps3, 2015-02-04:
My workaround was to simply use avrdude to perform the upload. It works just fine.

You need to reset the Pro Trinket first and wait for the first fade-up-down of LED indicating it's in bootloader mode, as per usual of course. Then, assuming you're in the top level directory of your `ino` project:

`$ avrdude -c usbtiny -p m328p -D -U flash:w:.build/protrinket3/firmware.hex:i`
*Note: the middle part of the pathname (i.e. `.build/*/firmware.hex`) will probably be different for you. You can probably either tab-complete while typing in that command line the first time.*

I've found that if I get an error with `avrdude` uploading, then the likely culprits are usually one of the following:
  1. My Mac has been offended by something in the USB chain and is ignoring that hub the Pro Trinket is plugged into, then I simply unplug the USB cable between the hub and the computer then back in again. I usually see no visual indication when this has happened, OS X is silent on these things, it seems.
  2. I was too slow and the Pro Trinket was out of bootloader/programming mode.

It took me a few tries to figure out the timing.

ivankravets, 2015-03-01:
Dear @simon-whitehead and @jps3,

Could I ask you to try it with [PlatformIO](http://platformio.org)? It has support for Adafruit's Flora & Trinket, but I haven't tested uploading yet (I don't have these boards). 

Thanks in advance!

closed 2017-11-27 by jps3