amperka/ino #232: ino build error

pekkert, 2015-02-16:
when i try to build i get this error?

ino build
Traceback (most recent call last):
  File "/usr/local/bin/ino", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 76, in main
    args.func(args)
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 284, in run
    self.setup_flags(args)
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 151, in setup_flags
    mcu = '-mmcu=' + board['build']['mcu']
KeyError: 'mcu'

malcolmsparks, 2015-02-22:
I get this error too 

Platform: Arch Linux
Installed via make install from ino git
HEAD=f23ee5cb14edc30ec087d3eab7b301736da42362

malcolmsparks, 2015-02-22:
From ino clean, which works

```
[malcolm@tulkas ard]$ ino build
Searching for Board description file (boards.txt) ... found multiple: 
  - /usr/share/arduino/hardware/arduino/avr/boards.txt
  - /usr/share/arduino/hardware/arduino/sam/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.6.0 (1.6.0)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/avr/cores/arduino
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/avr/variants
Searching for Arduino standard libraries ... /usr/share/arduino/libraries
Searching for make ... /usr/bin/make
Searching for avr-gcc ... /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /usr/share/arduino/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /usr/share/arduino/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /usr/share/arduino/hardware/tools/avr/bin/avr-objcopy
Traceback (most recent call last):
  File "/usr/local/bin/ino", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/site-packages/ino/runner.py", line 76, in main
    args.func(args)
  File "/usr/local/lib/python2.7/site-packages/ino/commands/build.py", line 284, in run
    self.setup_flags(args)
  File "/usr/local/lib/python2.7/site-packages/ino/commands/build.py", line 151, in setup_flags
    mcu = '-mmcu=' + board['build']['mcu']
KeyError: 'mcu'
```

malcolmsparks, 2015-02-22:
Do you have a  $HOME/.inorc file ? If so, try moving it away and retrying - I suspect the boards.txt it's finding has changed

malcolmsparks, 2015-02-22:
I fixed this on my own environment by downgrading the Arch AUR package of arduino from 1.6 to 1.0.6

pekkert, 2015-02-23:
Ok ill try it..

pekkert, 2015-02-24:
I did not have a .inorc file so i'll try to downgrade arduino.

pekkert, 2015-02-24:
I downloaded arduino 1.0.6 and installed it under /usr/local/share ( thats where ino looks for it )
Then i tried to do a 'ino build' under my test project, but i got the same error.

its on a raspberry pi running rapsbian..

pekkert, 2015-02-24:
i have looked at the boards.txt and every board in that file all have a mcu line like this..

mega2560.build.mcu=atmega2560

jgarbers, 2015-03-11:
Looks like this could be the same as my issue #228.  I wrote up a temporary workaround in that post; maybe that might help you.

Kr4ntz, 2015-07-27:
I cannot downgrade the IDE to 1.0.6 because some specific functions won´t work well on 1.0.X

Is there any solution to this error? here it is.

pi@vfa ~/vfa $ ino build -m mega
Searching for Board description file (boards.txt) ... found multiple: 
  - /usr/share/arduino/hardware/arduino/boards.txt
  - /usr/share/arduino/hardware/arduino/sam/boards.txt
  - /usr/share/arduino/hardware/arduino/avr/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.6.0 (2:1.6.0)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/avr/cores/arduino
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/avr/variants
Searching for Arduino standard libraries ... /usr/share/arduino/libraries
Searching for make ... /usr/share/arduino/hardware/tools/avr/bin/make
Searching for avr-gcc ... /usr/share/arduino/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /usr/share/arduino/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /usr/share/arduino/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /usr/share/arduino/hardware/tools/avr/bin/avr-objcopy
Traceback (most recent call last):
  File "/usr/local/bin/ino", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 76, in main
    args.func(args)
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 284, in run
    self.setup_flags(args)
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 151, in setup_flags
    mcu = '-mmcu=' + board['build']['mcu']
KeyError: 'mcu'

pomadomaphin, 2015-12-03:
I know this is a little bit of a late reply, but the problem comes from having spaces in the board file you are using. 

This is a problem because the python build file (build.py) is using the exact strings as dictionary keys.

you can remove the white space in the files that ino is pointing to, or you can modify the python build file to remove white space from all of the board related strings.

MalcolmMielle, 2016-07-03:
Remove the space did not solve the problem on Fedora 23