amperka/ino #234: attiny

mfulz, 2015-02-17:
Is it possible to program an attiny like described here: http://highlowtech.org/?p=1695

I couldn'd find any example or explanation on how to do this.

ajfisher, 2015-02-18:
So I've managed to get half way through this - I can get ino to compile but having a different arduino install and then using arduino-dist in my ino.ini.

The bit that I'm literally working through at the moment is how to change the command that AVRDUDE gets in order to be able to flash the chip.

JMAN2018, 2018-02-26:
HI GUYS. I HAVE A LDR light sensor I would like to use to turn on a light bulb through a attin85. when I upload the sketch to the Arduino it works fine. but when I upload to the attiny, I can't get it to work. am I'm wiring it wrong? when I put my signal from my ldr sensor to AO and 2 to the led it works on Arduino, but when I do that with the Attiny85 it doesn't do anything. The sketch did upload correctly though so it has to be me wiring it wrong or I need to change the code maybe to cater to the attiny85?

my sketch is 
int LDR = 0;    
int LDRValue = 0;      

void setup()   

{     

pinMode(2, OUTPUT); 

void loop()

  {     

LDRValue = analogRead(LDR);      

delay(50);        

      if (LDRValue < light_sensitivity)        

{         

digitalWrite(2, HIGH);      

}

    else       

{        

digitalWrite(2, LOW);       

}   

}