amperka/ino #236: Can't get ino to work on Arch on RaspberryPI

douglagm, 2015-02-25:
Hi,

I am trying to get ino working on a RaspberryPI using Arch and Arduino 1.6.0. I have installed ino using pip2. I am getting the following error:

Searching for Board description file (boards.txt) ... /usr/share/arduino/hardware/arduino/boards.txt
Traceback (most recent call last):
  File "/usr/bin/ino", line 6, in <module>
    main()
  File "/usr/lib/python2.7/site-packages/ino/runner.py", line 64, in main
    e.process_args(args)
  File "/usr/lib/python2.7/site-packages/ino/environment.py", line 263, in process_args
    all_models = self.board_models()
  File "/usr/lib/python2.7/site-packages/ino/environment.py", line 204, in board_models
    subdict[key] = {}
TypeError: 'str' object does not support item assignment

I have found references to this same issue in #111, #214, #226; How do I resolve this issue?

ivankravets, 2015-03-01:
`ino` doesn't support Arduino IDE > 1.5.0. I would propose you to try [PlatformIO](http://platformio.org). You don't need to install any IDEs, tool chains or etc. PlatformIO has all stuff from the "box" and it works on the card-sized PC, like Raspberry Py, BeagleBone, CubieBoard and etc.

scottdarch, 2015-03-31:
I've started a fork of ino (http://www.archxs.com/Arturo/) that works with the latest Arduino IDE. Let me know if this will work for your use case.