amperka/ino #240: Can't build; boards.txt not found after upgrading to Arduino 1.6.1

jgarbers, 2015-03-11:
It looks like the new version of the IDE moved the boards.txt file.  I get the following error when building:

    Board description file (boards.txt) not found. Searched in following places: 
      - /Applications/Arduino.app/Contents/Resources/Java/hardware/**
      - /usr/local/share/arduino/hardware/**
      - /usr/share/arduino/hardware/**

With the new version of the IDE, the file is found at

    /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt

scottdarch, 2015-03-13:
Seeing this as well.

scottdarch, 2015-03-20:
No commits to this project for a year now. I guess this is dead code.

scottdarch, 2015-03-30:
Figured it out. You have to ```ino clean``` first. The bug appears to be that the environment.pickle is not properly regenerated when a new version of the toolchain is used.

That said: once you clear this bug you'll probably run into the wifi library bug. You can grab [my pull request](https://github.com/amperka/ino/pull/242) if you want to get the fix.

jeremywrnr, 2015-03-30:
Excellent, I cloned your fork of the repo and it is working great! 

Thank you so much for following up :+1:

jgarbers, 2015-03-30:
I'm now using the fork from @scottdarch . On my Mac, however, "ino clean" didn't help the original problem of *boards.txt* not being found.  @scottdarch  and @jeremywrnr , what OS are you using that's working?  I have a minor fix that adds the new 1.6.1 location of *boards.txt* to the set of "guesses" for the Arduino distribution directory, but I don't want to bother posting a pull request for it if you're on Macs and have done something simpler to get it to work.

scottdarch, 2015-03-30:
I'm on a mac. OSx 10.10 (Yosemite).  Is your Arduino.app under /Applications/Arduino.app? And (sorry for the dumb questions) did you remember to make install from my fork after you cloned it?

Let me know. I want to get ino working again. If @amperka won't add me as a contributor I plan on starting an official fork.

jgarbers, 2015-03-30:
I'm also on Yosemite.  Somehow I had not noticed the *make install* step and just ran *python setup.py build* and *python setup.py install* explicitly... not sure why that would have mattered, but I seem to be working fine now after doing "make install".   Only thing I had to do in the makefile was change *python2* to *python*, since (I assume) I'm running the stock Yosemite-provided Python instead of anything from python.org or Homebrew.  

As pleased as I am that it's working, I don't understand how. Can you explain how *ino* is doing this:

    Searching for Board description file (boards.txt) ... found multiple: 
      - /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt
      - /Applications/Arduino.app/Contents/Java/hardware/arduino/sam/boards.txt

when the line in *environment.py* appears to be looking elsewhere:

    if platformSystem == 'Darwin':
        arduino_dist_dir_guesses.insert(0, '/Applications/Arduino.app/Contents/Resources/Java')

If you do decide to fork *ino* I'll be glad to offer whatever feedback and help I can. Thanks!

scottdarch, 2015-03-30:
The dist dir is the base path. From there the script searches under 'hardware/**' for all boards.txt files. These contain the information to find the proper platform.txt and programmer.txt files based on the board selected (i.e. for uno 'hardware/arduino/avr/platform.txt' but for due 'hardware/arduino/sam/platform.txt').

There is also the concept of 'vendor' that I haven't tested to see if ino is supporting. the 'arduino' under hardware is considered the hardware vendor. Theoretically it should work if you also had, say, 'hardware/ladyada/intel/boards.txt'

If I do fork I'm interested in getting first-class raspberry pi support possibly using [wiringPi](http://wiringpi.com/) or by building my own bcm toolchain. My dream is to be able to write-once-run-on-ar-or-pi.

jgarbers, 2015-03-30:
Sorry to muddy the waters, but my report that things were working was premature. I had apparently installed *ino* using *pip*, and I think I was running that version rather the one built from your fork.  After a *pip uninstall ino* and a *sudo make install* from your fork, I'm getting the original error:

    Searching for Board description file (boards.txt) ... FAILED
    Board description file (boards.txt) not found. Searched in following places: 
      - /Applications/Arduino.app/Contents/Resources/Java/hardware/**
      - /usr/local/share/arduino/hardware/**
      - /usr/share/arduino/hardware/**

I did do an *ino clean* per your earlier instructions (the above error came from the command *ino list-models*, which doesn't even need a project). I added a couple of print statements to make sure I was running from your source, and they do show up.

Are you using the Arduino.app "for Java 6 (recommended)" or "for Java 7 (experimental)"? Maybe there's a difference there?

I'm going to reinstall Arduino 1.6.1 for Java 6 from the download to see if there's some difference. Does your copy have the path */Applications/Arduino.app/Contents/Resources/Java/hardware/* ? Mine doesn't; will check against the newly-reinstalled one and report.

jgarbers, 2015-03-30:
Okay, I've reinstalled Arduino 1.6.1 for Java 6 and things appear to be working fine. The "experimental" 1.6.1 for Java 7 is no longer available for download, but I'm guessing that's what I had installed that caused all the trouble.  Given that @scottdarch and @jeremywrnr both reported the same problem, perhaps the *ino clean* step was enough for you guys (using the Java 6 version, I'd guess?) but not for me with the weird one.  Will be glad to close the ticket if everybody's good now.

scottdarch, 2015-03-30:
I'd leave the ticket open since the fix is not pulled into the repo. 

I got ahold of the amperka guys and they say they are too busy to maintain this tool. I am going to create a new fork. I'll try to work through some of these issues in the new repo once I get it setup.

Anyone have ideas for the name of the new fork? I'm thinking of calling it arturo and renaming the python command to ano. I want to call it something different so it's clear when someone is using ino versus the fork.

jeremywrnr, 2015-03-30:
For names: I like `arCLIno - the arduino CLI`, and you could call the python command `arc`.

However, if you are going to be the new maintainer, you should have final say on naming!

scottdarch, 2015-03-30:
Thanks for the suggestion Jeremy but arc is overloaded from iOS (automatic reference counting) and I do want arturo (there I used it; it's official ;-) to also support pi and other hardware in the future.

jeremywrnr, 2015-03-30:
Not a problem, I like that name too :+1:

jgarbers, 2015-03-30:
"arturo" reminds me of Toscanini... and for an Arduino build tool, I think a reference to a great Italian conductor is entirely appropriate (whether intended or not!):

http://en.wikipedia.org/wiki/Arturo_Toscanini

scottdarch, 2015-03-30:
It was actually named after the last contributor to the Arduino 3p IDE repo on github until now. Now it is named after Arturo Toscanini! Thanks @jgarbers.

jgarbers, 2015-03-30:
A conductor's baton would make a nice project icon if you were looking for one!

scottdarch, 2015-03-30:
[Introducing Arturo](https://github.com/scottdarch/Arturo)
Lots to be done. Let me know if there are any issues using ano along side ino.

chrisspen, 2015-11-26:
Has anyone tested this in 1.6.6? The format of boards.txt seems to have changed as well, and ino can no longer recognize it, even after I fix the location with a symlink. Running `ino build` gives me the error:

    Traceback (most recent call last):
      File "/usr/local/bin/ino", line 6, in <module>
        main()
      File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 76, in main
        args.func(args)
      File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 285, in run
        self.make('Makefile.sketch')
      File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 215, in make
        ret = subprocess.call([self.e.make, '-f', makefile, 'all'])
      File "/usr/lib/python2.7/subprocess.py", line 522, in call
        return Popen(*popenargs, **kwargs).wait()
      File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
        errread, errwrite)
      File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
        raise child_exception
    OSError: [Errno 2] No such file or directory

Noddegamrah, 2017-08-17:
August 2017, Arduino IDE 1.8.3. Still not working. Did anyone fork this project and still support it?

chrisspen, 2017-08-18:
This project is dead. I'd recommend using Arduino-Makefile, which provides the same features and is still actively maintained.

MauroFoti, 2019-06-18:
While installing `arduino-makefile`, I've found a solution. I noticed that togheter with `arudino-mk` it would install the following packages `arduino-core avr-libc avrdude binutils-avr gcc-avr libftdi1`, arduino-core seems particularly interesting, and after installing them it worked