amperka/ino #241: compilations in parallel fail

amlwwalker, 2015-03-19:
I have been doing research on using ino to compile code that is sent to a server from a web application. Everything works perfectly when I compile from one request however parallel compilations fail. 

Also even when one compilation is running the resources it uses are insane. Has anyone else checked this? I am running the builds inside docker containers so that I can compile programs in parallel, however beginning the compilation destroys the resources!
I can start as many docker containers as I like, it seems that building a simple blink example is unbelievably expensive.
I am running on an AWS m3.medium instance and this is htop when I start the compilation.

[image]

scottdarch, 2015-03-31:
Hello, I'm continuing ino development on a fork here: http://www.archxs.com/Arturo/ (the ino maintainers have blessed this fork)

Your issues sounds interesting. I suspect it's cause by overly aggressive filesystem traversal when searching for library dependencies.

First try the forked version with the latest Arduino release (1.6.1 or later) and let me know if the problem continues by opening an issue in the Arturo project.

cheers,
-sd

ches, 2015-04-22:
> Hello, I'm continuing ino development on a fork here: http://www.32bits.io/Arturo/ (the ino maintainers have blessed this fork)

That's good news, and thanks for taking up maintenance effort. Although I wonder why the change in project and executable name if the fork is blessed... but probably that's best discussed elsewhere.

Where did this blessing happen? It would be great if Ino developers could at least update the GitHub README and ideally inotool.org to warn that the project isn't being maintained at this time, there are still lots of issues and pull requests being opened on this repository and it's a shame for peoples' efforts to be wasted on a version that's unlikely to take merges.

scottdarch, 2015-04-22:
I've started an FAQ with an answer: https://github.com/scottdarch/Arturo/wiki/FAQ

ches, 2015-04-22:
That's a great place to start, thanks @scottdarch!