amperka/ino #251: Argument list too long

raivof, 2015-05-16:
1. I have just installed ino through "git clone / make install".

2. created empty project using "ino init"
3. and, by executing "ino build", I get error:

`Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.6.3 (1.6.3)
avr-gcc: error trying to exec 
'/home/user1/apps/arduino/hardware/tools/avr/bin/../libexec/gcc/avr/4.8.1/cc1plus': execv: Argument list too long
make: *** [.build/uno/src/sketch.d] Error 1
Make failed with code 2`

OS: Ubuntu-based Linux Mint

lucciano, 2015-05-24:
Same here, with 1.6.4.

bao3, 2015-06-05:
same error .

Arduino IDE 1.6.4

dimascrocco, 2015-06-08:
arghhhhh... same thing with 1.5.6 =/

bremby, 2015-07-04:
Just tried running it with the verbose option. The problem really is what the error describes. It lists all possible include paths to the compiler, and it won't even fit to my terminal buffer to display all of it.

allonhadaya, 2015-07-22:
I can confirm this issue as well. Ino: f23ee5cb14edc30ec087d3eab7b301736da42362, Ubuntu: 14.04, Arduino: 1.6.5

allonhadaya, 2015-07-22:
I would guess that this line is problematic: https://github.com/amperka/ino/blob/f23ee5cb14edc30ec087d3eab7b301736da42362/ino/commands/build.py#L224

josh-kaplan, 2015-07-28:
Apparently it creates a 15MB Makefile.deps. I changed line #224 to `recursive=False` and got it to build on OS X. Haven't tried to upload it to a board yet.

chrisspen, 2015-11-26:
@josh-kaplan, That fixed it for me on Ubuntu 14.04! Thanks. I now have 1.6.6 installed. See my list of other necessary fixes in https://github.com/amperka/ino/issues/260.

limikael, 2017-11-14:
Getting this error... :(

Arduino version: 1.8.5
Os version: Ubuntu 16.10