amperka/ino #252: No module named glob2

uocxp, 2015-06-03:
I have just installed the inotool. when i run "ino --help" i get:
Traceback (most recent call last):
  File "/usr/local/bin/ino", line 3, in <module>
    from ino.runner import main
  File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 23, in <module>
    import ino.commands
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/__init__.py", line 5, in <module>
    from ino.commands.build import Build
  File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 16, in <module>
    from ino.environment import Version
  File "/usr/local/lib/python2.7/dist-packages/ino/environment.py", line 19, in <module>
    from glob2 import glob
ImportError: No module named glob2

any help ?
I'm using Ubuntu 14.04

bao3, 2015-06-05:
I had the same issue ,but you could try to fix this by type in Terminal : 

pip install glob2

good luck

uocxp, 2015-06-07:
glob2 is already installed :/

r0estir0bbe, 2015-07-22:
The Ubuntu package `glob2` is different than the required `glob2` (a Python library). To install it, you can install `python-pip` and then `pip` to install it:

```bash
sudo apt-get install python-pip
sudo pip install glob2
```

docdawning, 2015-12-31:
Thanks fellas. That worked for me.

ipinak, 2016-01-07:
You just need to run `pip install -r requirements.txt` in case you have built this from source. If you install it through `pip` you shouldn't have this problem.