amperka/ino #254: 0.3.6 : No project found in this directory

J-Dunn, 2015-07-16:
I've just installed ino-0.3.6 on Fed20. It falls at the first hurdle:
```
$ ino preproc  grbl_sketch.ino
No project found in this directory

bash-4.2$ ls -ail 
total 12
1483071 drwxrwxr-x.  2 stm stm 4096 Jul  5 08:39 .
1483062 drwxrwxr-x. 13 stm stm 4096 Jul 15 13:36 ..
1487057 -rw-rw-r--.  1 stm stm  447 Jul  5 08:39 grbl_sketch.ino
0.3.6 on Fedora .
```

Not encouraging.

GilDev, 2015-08-08:
I may be wrong but I think you should put your source files in a "src" folder.

GTpyro, 2017-06-24:
@GilDev yes, you are correct. ino expects the following structure:
- lib
- src
  - sketch.ino

Otherwise, ino will fail. I recommend this issue be closed.