amperka/ino #255: Is this project still being maintained?

stormbytes, 2015-08-19:
Doesn't seem to be compatible with the latest Arduino IDE (and toolchain).

I'm on Mac 10.7.

So far the alternative has been to use Arduino's own very clunky and difficult to use CLI. Please someone update Ino !!

Iniesta8, 2015-10-23:
Yes, it seems to be outdated and not compatible with the latest Arduino IDE and also Mac OS X El Captain...pls maintain it!

em, 2016-02-17:
@amperka Would you mind finding a new maintainer? There's 35 open pull requests ATM and no commits for 2 years.

chrisspen, 2016-02-17:
FYI, the [ano](https://github.com/scottdarch/Arturo) project is a ino fork for just that purpose.

Iniesta8, 2016-02-18:
You can also use Platform.io (https://github.com/platformio/platformio) which is really awesome.

watsoncj, 2016-02-24:
@nailxx @12qu @chrisspen and any interested parties,

Ino is such a cool toolkit. Great work. The community is begging to use and contribute to it.

The inotools name has a lot of recognition. I'm worried that we are fragmenting the community by renaming the project (ano). This results wasting valuable contributions.

This project belongs under a dedicated organization where the community can continue to move it forward.

- Appoint some project maintainers.
- Create a dedicated inotool organization.
- Migrate this project to the organization. (Ideally using Github's transfer function to keep issues and setup the proper redirects)

There are many issues that have been fixed multiple times. Tons of work out on branches that just needs tested and merged. 

Lets open this project up to some active stewards so that the community can move it forward!

chrisspen, 2016-02-25:
That would require the project creator's permission, and he's AWOL, so
that's not going to happen. I'd recommend helping out with Ano. That Dev
responds to issue submissions and is actively looking for help to port the
code to the most recent Arduino release.

mogorman, 2016-02-25:
id second looking at plaformio.org its awesome