amperka/ino #260: Compatibility with Arduino 1.6.6?

chrisspen, 2015-11-26:
After installing Arduino 1.6.6, `ino build` fails with the error:

    Traceback (most recent call last):
      File "/usr/local/bin/ino", line 6, in <module>
        main()
      File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 76, in main
        args.func(args)
      File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 285, in run
        self.make('Makefile.sketch')
      File "/usr/local/lib/python2.7/dist-packages/ino/commands/build.py", line 215, in make
        ret = subprocess.call([self.e.make, '-f', makefile, 'all'])
      File "/usr/lib/python2.7/subprocess.py", line 522, in call
        return Popen(*popenargs, **kwargs).wait()
      File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
        errread, errwrite)
      File "/usr/lib/python2.7/subprocess.py", line 1327, in _execute_child
        raise child_exception
    OSError: [Errno 2] No such file or directory

Inspecting the code, it's trying to call the binary `/usr/share/arduino/hardware/tools/avr/bin/make` which no longer seems to exist in 1.6.6.

chrisspen, 2015-11-26:
I worked around this error by symlinking the standard `/usr/bin/make` into that hard-coded location. However, then I received the error:

    No such file or directory: /usr/share/arduino/hardware/arduino/cores/arduino

This error is reported in other issues, and the fix is to symlink:

    sudo ln -s /usr/share/arduino/hardware/arduino/avr/cores /usr/share/arduino/hardware/arduino/
    sudo ln -s /usr/share/arduino/hardware/arduino/avr/boards.txt /usr/share/arduino/hardware/arduino

But now I get the error:

    Traceback (most recent call last):
      File "/usr/local/bin/ino", line 6, in <module>
        main()
      File "/usr/local/lib/python2.7/dist-packages/ino/runner.py", line 64, in main
        e.process_args(args)
      File "/usr/local/lib/python2.7/dist-packages/ino/environment.py", line 263, in process_args
        all_models = self.board_models()
      File "/usr/local/lib/python2.7/dist-packages/ino/environment.py", line 204, in board_models
        subdict[key] = {}
    TypeError: 'str' object does not support item assignment

chrisspen, 2015-11-26:
It seems the format of boards.txt has changed slightly, causing ino's parsing of the file to fail. I fixed this error by patching line 204 in environment.py with:

    for key in multikey[:-1]:
        if key not in subdict:
            subdict[key] = {}
        if not isinstance(subdict[key], dict):
            subdict[key] = {}
        subdict = subdict[key]

I also had to add a symlink for the variants folder:

    sudo ln -s /usr/share/arduino/hardware/arduino/avr/variants /usr/share/arduino/hardware/arduino/variants

However, now I'm stuck on the error:

    avr-gcc: error trying to exec '/usr/share/arduino/hardware/tools/avr/bin/../libexec/gcc/avr/4.8.1/cc1plus': execv: Argument list too long

chrisspen, 2015-11-26:
The fix for this last error is mentioned in https://github.com/amperka/ino/issues/251. Setting recursive=False in recursive_inc_lib_flags of build.py is the problem, as it generates a list of include paths too long for avr-gcc.

Please let me know if I should submit a patch. I still have an outstanding pull request, so if this package is no longer being maintained, I won't waste my time.

Gasman2014, 2016-05-05:
Did all your amendments result in a functional copy of ino? I am asking to see if it is worth doing?

chrisspen, 2016-05-06:
I've given up on ino. I recommend using ano instead, which is a more recently maintained fork.

closed 2016-05-06 by chrisspen

Gasman2014, 2016-05-06:
Shame but thanks - some config problems with ano too ;(

John