amperka/ino #270: please expand quick start to summarize initial installation

archenemies, 2016-05-19:
Hello! I was just looking at your quick start guide/tutorial. I haven't actually tried the steps, but I noticed that your output of `ino build`, the first step, contains the line:

    Searching for Board description file (boards.txt) ... /usr/local/share/arduino/hardware/arduino/boards.txt

This seems to imply that I should have this `boards.txt` file somewhere on my system. Please update quick start with a few sentences explaining what other packages have to be installed before users can begin the tutorial. I think this will make it more accessible to beginners. Thank you!

nophead, 2016-05-19:
If you look at the requirements section of the README you will see that you need the Arduino IDE installed. That is where the boards.txt comes from.

archenemies, 2016-05-19:
Thanks! I think it would be good to put that information in the tutorial. Just a few lines or a link about installing the IDE and whatever configuration is needed beyond that. From what I have read, your project would provide a more useful interface than the IDE for a user like myself, so I'm not necessarily interested in learning about the IDE beyond what is needed to use `ino`. Does that make sense?

Just for context, I have untarred something which calls itself the Arduino IDE, and I ran `./install.sh`, but the only location for `boards.txt` on my filesystem is in the tarball directory, nothing in `/usr/share/`... So that's how little I know about what "Requirements: Arduino IDE" means.

Thank you!

nophead, 2016-05-19:
All I needed to do was
```
sudo apt-get install arduino
```

archenemies, 2016-05-19:
OK, it's not in the official repos of my distribution.

But if you did `apt-get install arduino`, it would be useful to put that line in the tutorial as it tells me a bit where to look. Thanks.

archenemies, 2016-05-19:
How's this:

    --- README.rst	2016-05-19 16:29:36.809358789 -0700
    +++ README.rst.1	2016-05-19 16:30:28.900644240 -0700
    @@ -57,7 +57,7 @@
     ============

     * Python 2.6+
    -* Arduino IDE distribution
    +* Arduino IDE distribution (``apt-get install arduino``)
     * ``picocom`` for serial communication

     Limitations

archenemies, 2016-05-20:
Or this:

    --- quickstart.rst.old	2016-05-20 11:28:32.698244860 -0700
    +++ quickstart.rst	2016-05-20 11:43:11.259839158 -0700
    @@ -4,6 +4,21 @@

     Learn how to work with ino in few minutes.

    +
    +Prerequisites
    +-------------
    +
    +Make sure you have first installed Ino and its prerequisites as
    +described in the `README <http://inotool.org/>`_.
    +
    +Note that the Arduino IDE should be installed from your distribution's
    +packaging system (e.g. ``apt-get install arduino``), so that
    +``boards.txt`` can be found in a standard location under
    +``/usr/share/`` or ``/usr/local/share/``. The `Arduino website
    +binaries <https://www.arduino.cc/en/Main/Software>`_ provide an
    +"in-place" installation, which won't work for us.
    +
    +
     Creating a project
     ------------------

shaleh, 2016-06-03:
Looks good. Commit it :+1:

archenemies, 2016-06-04:
Hey thanks. By the way is Inotool still being used? This URL

http://playground.arduino.cc/Learning/CommandLine

says "The popular command line builder, Ino (available at http://inotool.org/) has not seen active development in some time and no longer works with newer versions of the Arduino IDE". If that is true, I think it should be in the Quick Start Guide as well...

shaleh, 2016-06-04:
Yeah, it looks like ino is dead. https://github.com/scottdarch/Arturo is meant to be the continued fork. It ships a command called 'ano'.

archenemies, 2016-06-04:
Did the maintainer just die or something? How odd...